YDLIDAR/YDLidar-SDK #1: read lidar data from python

ggjj11, 2020-03-22:
Hello thank you for the very interesting lidar products which you offer!

I bought the X2 lidar and tried to read its data using python and windows.
I tried to run the build and install steps (https://github.com/YDLIDAR/YDLidar-SDK/blob/master/doc/howto/how_to_build_and_install.md)

The following command failed due to some cmake path error 

> pip install .

running 

>python setup.py build
>python setup.py install

also fails in the build step (maybe because the previous VS compilation failed in the ALL_BUILD step?).
error: [Errno 2] No such file or directory: 'build\\temp.win-amd64-3.7\\Release\\python/ydlidar.py'

(here also the slash looks suspicous, but changing it to a backslash did not help).

I don't know what goes wrong here, but I guess I will have to return the lidar within the first two weeks if this is not fixed quickly.

yangfuyuan, 2020-03-23:
Hi,
For windows requires Visual Studio 2015 or later.

ggjj11, 2020-03-23:
I used visual studio 2019

yangfuyuan, 2020-03-23:
Make sure the python and swig environment variables are set correctly. If you still get errors, show me your build failure log.

ggjj11, 2020-03-23:
Thank you! After installing swig, I was able to compile your SDK an run some python samples, where I needed to adapt the baudrate:
> laser.setlidaropt(ydlidar.LidarPropSerialBaudrate, 115200)
as well as 
> laser.setlidaropt(ydlidar.LidarPropSingleChannel, True);
Is this all what is needed to adapt?

yangfuyuan, 2020-03-23:
Also you need to set  SampleRate prpperty
```
   laser.setlidaropt(ydlidar.LidarPropSampleRate, 3);
```
For more details, please refer to SDK documentation

ggjj11, 2020-03-23:
Thank you for your help. I will have a look.

closed 2020-03-23 by ggjj11

sureshkannaiyan, 2024-02-22:
Hello There,
While executing this command, it throws error
cmake -G "Visual Studio 16 2019" "-DCMAKE_TOOLCHAIN_FILE=C:/Users/sures/vcpkg/scripts/buildsystems/vcpkg.cmake"

ERROR:

C:\Users\sures\YDLidar-SDK\build>cmake -G "Visual Studio 16 2019" "-DCMAKE_TOOLCHAIN_FILE=C:/Users/sures/vcpkg/scripts/buildsystems/vcpkg.cmake"
CMake Warning:
  No source or binary directory provided.  Both will be assumed to be the
  same as the current working directory, but note that this warning will
  become a fatal error in future CMake releases.

CMake Error at CMakeLists.txt:3 (project):
  Generator

    Visual Studio 16 2019

  could not find any instance of Visual Studio.

-- Configuring incomplete, errors occurred!

Any suggestion to resolve this issue would be of great help

peterhczaja, 2024-12-21:
The Ydlidar X2 does NOT work on the Raspberry Pi under Python3