YDLIDAR/YDLidar-SDK #2: read lidar data with pure python

ggjj11, 2020-03-22:
I found several (**pure** python) github projects which read data from the lidars using the serial package, but unfortunately none for the X2:

- https://github.com/lakshmanmallidi/PyLidar3 (almost working for the X2, but crashing after 10 seconds, probably because there is some X4 specifics)
- https://github.com/SweiLz/PyYDLidar
- https://github.com/nesnes/YDLidarX2_python (unfortunately not working for me)

Could you maybe check the PyLidar3 project and post the needed adaptions for getting the X2 to work with it. This would help me a lot.

Maybe you could also provide a simple python package using the serial package to read the lidar data? This would be much easier than your current solution.

yangfuyuan, 2020-03-26:
See Issue #1

closed 2020-03-26 by yangfuyuan

ggjj11, 2020-03-26:
I think this would still be beneficial. Alternatively you could create a conda package, such that it is easier to install the sdk. There you can also distribute binaries.

yangfuyuan, 2020-03-26:
Thanks for your good suggestions, we will consider using conda to create an installation package, and release the binary