YDLIDAR/YDLidar-SDK #3: Set the scanning frequency of the X2 lidar

ggjj11, 2020-03-23:
Thank you again for your nice product and software! I am trying to get the X2 do what I want :)

I noticed that the X2 rotates extremely fast when plugging in the USB-C cable (and not running a program in the beginning) - actually so fast that I wonder whether this is healthy for the X2.
However, when starting a python script or (one of the provided programs in the SDK) which reads lidar data, the rotation speed is reduced to roughly 6Hz +/-1 Hz (standard deviation).
I tried increasing/decreasing the speed of rotation via increasing the ROTATION_FREQ, e.g. to 10 [Hz]

>laser.setlidaropt(ydlidar.LidarPropScanFrequency, ROTATION_FREQ);
> ... scan = ydlidar.LaserScan()

but having a look at two successive 'scan.stamp' times (in nanoseconds) only showed that I cannot really adapt the scanning frequencies like this: The scanning frequencies remained 6Hz +/-1 Hz (standard deviation).

Below you find the example script which I adapted from the example in this repository:

''''
import os
import ydlidar
import time
import sys
import matplotlib.pyplot as plt
import matplotlib.animation as animation
import numpy as np

ROTATION_FREQ=10 # in Hertz

fig = plt.figure()
fig.canvas.set_window_title('YDLidar LIDAR Monitor')
ax = plt.subplot()
ax.autoscale_view(False,False,False)
ax.grid(True)
ports = ydlidar.lidarPortList();
port = "com3" 
for key, value in ports.items():
    port = value;
    
laser = ydlidar.CYdLidar();
laser.setlidaropt(ydlidar.LidarPropSerialPort, port);
laser.setlidaropt(ydlidar.LidarPropSerialBaudrate, 115200) #115200
laser.setlidaropt(ydlidar.LidarPropLidarType, ydlidar.TYPE_TRIANGLE); 
laser.setlidaropt(ydlidar.LidarPropDeviceType, ydlidar.YDLIDAR_TYPE_SERIAL);
laser.setlidaropt(ydlidar.LidarPropScanFrequency, ROTATION_FREQ);
laser.setlidaropt(ydlidar.LidarPropSampleRate, 3);
laser.setlidaropt(ydlidar.LidarPropSingleChannel, True);
scan = ydlidar.LaserScan()

scan_times=[]

def animate(num):
    
    r = laser.doProcessSimple(scan);
    if r:
        angle = []
        ran = []
        intensity = []
        scan_times.append(scan.stamp) #begin of scan, returned in nanoseconds 
        for point in scan.points:
            angle.append(point.angle);
            ran.append(point.range);
            intensity.append(point.intensity)
        print(len(intensity))
        angle = -np.array(angle) #minus because the ydlidar increases the angle clock-wise
        xs=np.array(ran)*np.cos(angle)
        ys=np.array(ran)*np.sin(angle)
        #timestamps=np.linspace(scan_times[-1]-scan_times[-2],num=xs.shape[0])
        ax.clear()
        ax.axis("equal")
        ax.set(xlim=(-3, 3), ylim=(-3, 3))
        ax.scatter(xs, ys, c=intensity, cmap='hsv', alpha=0.95, marker ='x')
        print('freq. in Hertz', 1/(np.diff(scan_times)*1e-9))
        plt.plot([0],[0], "go")

ret = laser.initialize();
if ret:
    ret = laser.turnOn();
    if ret:
        ani = animation.FuncAnimation(fig, animate, interval=10)
        plt.show()
    laser.turnOff();
laser.disconnecting();
plt.close();

''''

This is maybe related to: https://github.com/YDLIDAR/ydlidar_ros/issues/1

yangfuyuan, 2020-03-26:
X2L cannot adjust the scanning frequency through sdk, it needs external pwm. 
 The serial port DRT of the lidar adapter board is connected to the M_SCTR of the lidar. The DTR has two states, high and low, and the frequency of the fixed lidar is only two states.

scholzol, 2022-06-23:
How can I stop the lidar (X2L) over the USB- adapter ? I couldn't find any command in the API. when I switsch off, the lidar rotates with that high frequency, which is really not healthy.

halcyon1945, 2022-10-14:
i saw datasheet.
x2 interface is only send to PC.
no input(receive).