YDLIDAR/YDLidar-SDK #4: turning on the laser works with ROS, but not python

kevindean, 2020-04-30:
Hello,

like the other person stated, thank you for your product (G2B), it's epic. However, I can only seem to get it to generate a point cloud from ROS; and visualize it in Rviz (other words, laser won't turn on from python environment; it begins to rotate at the set frequency and then the thread is terminated). I've provided a couple images to show this. Are there any next steps I should take in debugging this issue? Thank you for your time - Kevin

[image]
[image]
[image]

yangfuyuan, 2020-05-03:
Hi Kevin,
  G2B lidar has intensity, so it is necessary to set `LidarPropIntenstiy ` property to True.
eg:
    ```
    laser.setlidaropt(ydlidar.LidarPropIntenstiy, True);
    ```

kevindean, 2020-05-03:
Awesome! Works perfectly, thank you very much for the help.

Sincerely,
Kevin