YDLIDAR/YDLidar-SDK #5: How to create cshap project doesn't run as tutorial

HuuLinhDra, 2020-09-11:
I installed swig and cmake like your tutorial 
https://github.com/YDLIDAR/YDLidar-SDK/blob/master/doc/howto/how_to_create_a_csharp_project.md
But, when I ran cmake -DBUILD_CSHARP=ON .. -G "Visual Studio 15 2017 Win64" "-DCMAKE_TOOLCHAIN_FILE=C:\DEV\vcpkg\scripts\buildsystems\vcpkg.cmake"
in build/csharp folder didn't have any csharp code. 
Please check it again

yangfuyuan, 2021-01-30:
You can follow a sample setup:
- 1.You need to install [Swig.](http://www.swig.org/download.html)(first)
- 2.Set BUILD_CSHARP option to ON:

     `cmake -DBUILD_CSHARP=ON .. -G "Visual Studio 15 2017 Win64" "-DCMAKE_TOOLCHAIN_FILE=C:\DEV\vcpkg\scripts\buildsystems\vcpkg.cmake"`

ashfrog, 2022-05-13:
> 

Can you give me an example of how to use ydlidar_csharp.dll to get radar data?