YDLIDAR/YDLidar-SDK #8: Error, cannot retreive YDLidar health code: ffffffff

yoansj, 2021-06-06:
Hello,

I recently brought the X4 Lidar and tried to run it on my raspberry pi 3b+ with ubuntu 18.04 as requested in the guide.
When i plug my lidar and run the

oleorhagen, 2021-09-27:
I had similar issues actually, and it turned out that the unit just wasn't receiving enough juice from my powersupply. An upgrade fixed it!

slowrunner, 2022-10-16:
I have same issue with my X4 on Pi4 on Ubuntu 22.04 that I do not have on Ubuntu 20.04 
Same robot, same setup steps, runs /usr/local/bin/ydlidar_test fine on Ubuntu 20.04 and health code: ffffffff on Ubuntu 22.04.  

This is not a power supply issue.

(Power supply is separate 5.02v 2A regulated supply with nothing else but YDLIDAR X4 powered.)

kleen8, 2022-12-09:
i have the same issue with my X4, anybody solved this problem yet?

Boogiebox, 2023-05-05:
that is also my problem...
using ubuntu 20.04.06 LTS, after ubuntu 22.04.02 LTS
using a 5 Volt 3 Ampere powersupply

my results...
torsten@p70:~/YDLidar-SDK/startup$ tri_test

[0] ydlidar /dev/ttyS9
[1] ydlidar6 /dev/ttyUSB0
Please select the lidar port:1
Baudrate:
[0] 115200
[1] 128000
[2] 150000
[3] 153600
[4] 230400
[5] 460800
[6] 512000
Please select the lidar baudrate:6
Whether the Lidar is one-way communication [yes/no]:no
Please enter the lidar scan frequency[5-12]:5
[YDLIDAR] SDK initializing
[YDLIDAR] SDK has been initialized
[YDLIDAR] SDK Version: 1.1.7
[YDLIDAR] Lidar successfully connected /dev/ttyUSB0[512000]
[YDLIDAR] Error, cannot retrieve YDLidar health code: ffffffff
[YDLIDAR] Fail to get device information
[YDLIDAR] Lidar init success, Elapsed time 2666 ms
[YDLIDAR] Failed to start scan mode: ffffffff
Fail to start Unknown error

a another test was...
[0] ydlidar /dev/ttyS9
[1] ydlidar6 /dev/ttyUSB0
Please select the lidar port:1
Baudrate:
[0] 115200
[1] 128000
[2] 150000
[3] 153600
[4] 230400
[5] 460800
[6] 512000
Please select the lidar baudrate:6
Whether the Lidar is one-way communication [yes/no]:yes
[YDLIDAR] SDK initializing
[YDLIDAR] SDK has been initialized
[YDLIDAR] SDK Version: 1.1.7
[YDLIDAR] Lidar successfully connected /dev/ttyUSB0[512000]
[YDLIDAR] Lidar running correctly! The health status: good
[YDLIDAR] Lidar init success, Elapsed time 624 ms
[YDLIDAR] Start to getting intensity flag
[YDLIDAR] End to getting intensity flag
[YDLIDAR] Create thread 0xE9B93700
[YDLIDAR] Successed to start scan mode, Elapsed time 3562 ms
timeout count: 1
timeout count: 2
timeout count: 1
timeout count: 2
[YDLIDAR DEBUG] Thread [0xE9B93700] ready to cancel[0]
[YDLIDAR DEBUG] Thread [0xE9B93700] ready to cancel[0] time[0]
[YDLIDAR] Thread [0xE9B93700] has been canceled
[YDLIDAR] Failed to turn on the Lidar, because the lidar is [Device Failed].
Fail to start Unknown error

**any suggestions?**

Boogiebox, 2023-05-05:
this is the file...

ydlidar_ros2_driver_node:
  ros__parameters:
    port: /dev/ttyUSB0
    frame_id: laser_frame
    ignore_array: ""
    baudrate: 230400
    lidar_type: 1
    device_type: 0
    sample_rate: 9
    abnormal_check_count: 4
    fixed_resolution: true
    reversion: true
    inverted: true
    auto_reconnect: true
    isSingleChannel: false
    intensity: false
    support_motor_dtr: false
    angle_max: 180.0
    angle_min: -180.0
    range_max: 64.0
    range_min: 0.01
    frequency: 10.0
    invalid_range_is_inf: false

Rih348, 2023-05-05:
what type is your ydlidar

Boogiebox, 2023-05-08:
it is a ydlidar X4.

Boogiebox, 2023-05-10:
Finally it works.
I tried always with baudrate **230000**, but only **128000** works propper.

Now my _ydlidar.yaml:_
ydlidar_ros2_driver_node:
  ros__parameters:
    port: /dev/ydlidar   #ttyUSB0
    frame_id: laser_frame
    ignore_array: ""
    baudrate: 128000
    lidar_type: 1
    device_type: 6 # 0
    sample_rate: 9 # 10
    abnormal_check_count: 4
    fixed_resolution: true
    reversion: false # true
    inverted: true
    auto_reconnect: true
    isSingleChannel: false
    intensity: false
    support_motor_dtr: true
    angle_max: 180.0
    angle_min: -180.0
    range_max: 10.0
    range_min: 0.12
    frequency: 7.0
    invalid_range_is_inf: false

jaafar-omar, 2023-08-15:
Instead of using the ydlidar_ros_driver_node, try to use:
`roslaunch ydlidar_ros_driver X4.launch` and in X4.launch file modify it to your actual use.

lvxinliang, 2024-04-23:
我修改了`ydlidar_ros2_driver/params/ydlidar.yaml`文件如下，可以正常使用了
```
ydlidar_ros2_driver_node:
  ros__parameters:
    port: /dev/ttyUSB0
    frame_id: laser_frame
    ignore_array: ""
    baudrate: 115200
    lidar_type: 1   # 0:TYPE_TOF, 1:TYPE_TRIANGLE 2:TYPE_TOF_NET
    device_type: 0   # 0:YDLIDAR_TYPE_SERIAL, 1:YDLIDAR_TYPE_TCP
    sample_rate: 3
    abnormal_check_count: 4
    fixed_resolution: true
    reversion: true
    inverted: true
    auto_reconnect: true
    isSingleChannel: true
    intensity: false
    support_motor_dtr: true
    angle_max: 180.0
    angle_min: -180.0
    range_max: 64.0
    range_min: 0.01
    frequency: 10.0
    invalid_range_is_inf: false

```

ulufun, 2024-12-22:
> Hello,
> 
> I recently brought the X4 Lidar and tried to run it on my raspberry pi 3b+ with ubuntu 18.04 as requested in the guide. When i plug my lidar and run the test i get this output
> 
> [image]
> 
> Any help will be cool

you should check your usb ports. you may use an extra usb and than at the launch file you may need to change your usb port id. i had same issue. after that  i changed my dev/ttyusb1 to dev/ttyusb0 and it is solved.