YDLIDAR/YDLidar-SDK #16: build folder is not seen in the YDLidar-SDK

PLEX-GR00T, 2022-02-06:
git clone https://github.com/YDLIDAR/YDLidar-SDK.git
cd YDLidar-SDK/build
cmake ..
make
sudo make install

Trying to follow the instruction to build the SDK for ROS driver from here. https://github.com/YDLIDAR/ydlidar_ros_driver
And  ROS driver gives exact same error as mentioned "make cmake_check_build_system" Error. Even though I have download the SDK and did Python API installation. 

I am new to ROS too. Am i missing something for how to build SDK. 
is "cd YDLidar-SDK/build" is workspace or SDK has build folder I don't understand it.
Its basics, but give me some hints pls. 

Thank you.

Moha-med1, 2022-03-17:
Try `cd YDLidar-SDK/src` instead of `cd YDLidar-SDK/build`

thehamzan6, 2022-03-27:
I created the build folder and continued as described and all was good.