YDLIDAR/YDLidar-SDK #18: ydlidar timeout 1 timeout 2

weihaoysgs, 2022-04-18:
when i use ydlidar X4 , i use the serial tool to read ydlidar data, the lidar is  normal rotation.
but when i use the launch cmd to launch the ydlidar, i have the following issure, then the ydlidar is stop

Firmware version: 1.10
Hardware version: 1
Model: X4
Serial: 2020062200002209
[YDLIDAR INFO] Current Sampling Rate : 5K
[YDLIDAR INFO] Current Scan Frequency: 7.600000Hz
timout count: 1
timout count: 2
fhs_unlock: Removing LockFile
fhs_lock: creating lockfile:       9137

it is always timeout 1 timeout 2

Manfe07, 2022-05-31:
Same problem here.
When i use the Windows Software from the website [ydlidar.com](https://www.ydlidar.com)
But when I try to use the SDK sample under Ubuntu 20.04 I get the timeout count.

Ataturkcu, 2023-04-08:
same error. Is there any update?
__   ______  _     ___ ____    _    ____  
\ \ / /  _ \| |   |_ _|  _ \  / \  |  _ \ 
 \ V /| | | | |    | || | | |/ _ \ | |_) | 
  | | | |_| | |___ | || |_| / ___ \|  _ <  
  |_| |____/|_____|___|____/_/   \_\_| \_\ 

[0] ydlidar /dev/ttyS9
[1] ydlidar12 /dev/ttyUSB0
Please select the lidar port:1
Baudrate:
[0] 115200
[1] 128000
[2] 150000
[3] 153600
[4] 230400
[5] 460800
[6] 512000
Please select the lidar baudrate:6
Whether the Lidar is one-way communication [yes/no]:yes
[YDLIDAR] SDK initializing
[YDLIDAR] SDK has been initialized
[YDLIDAR] SDK Version: 1.1.6
[YDLIDAR] Lidar successfully connected /dev/ttyUSB0[512000]
[YDLIDAR] Lidar running correctly! The health status: good
[YDLIDAR] Lidar init success, Elapsed time 623 ms
[YDLIDAR] Start to getting intensity flag
[YDLIDAR] End to getting intensity flag
[YDLIDAR] Create thread 0xC25FF640
[YDLIDAR] Successed to start scan mode, Elapsed time 2060 ms
timeout count: 1
timeout count: 2
timeout count: 1
[YDLIDAR DEBUG] Thread [0xC25FF640] ready to cancel[0]
[YDLIDAR DEBUG] Thread [0xC25FF640] ready to cancel[0] time[0]
[YDLIDAR] Thread [0xC25FF640] has been canceled
[YDLIDAR] Failed to turn on the Lidar, because the lidar is [Device Failed].
Fail to start Unknown error

NiuCY, 2023-10-12:
Different types of YDLidar have different sampling frequencies. This error should be caused by a mismatch in the sampling frequency settings.Try another baudrate.

NiuCY, 2023-10-12:
> Different types of YDLidar have different sampling frequencies. This error should be caused by a mismatch in the sampling frequency settings.Try another baudrate.

x2: 115200
x4: 128000

arxidinakbar, 2023-10-30:
Could you help on how to use Viewer on Windows, after configurate the IP address I just can't open the website  : (