YDLIDAR/YDLidar-SDK #21: Include build directory in repo

micahreich, 2022-06-17:
The installation tutorial instructs you do ``cd`` into ``YDLidar-SDK/build`` but this directory does not exist in the repository upon cloning. For the sake of new users, add the empty build directory to the repository to circumvent any confusion, since nowhere in the installation instructions does it say to make a new directory.

opteron18, 2022-07-25:
i got the same problem.

JHPHELAN, 2022-09-05:
Same here.  Following advice of @micahreich above I created the /build directory, entered it and ran cmake with following warnings, and fatal errors:
```
ubuntu@LINOROBOT:~/YDLidar_SDK.ws/YDLidar-SDK$ mkdir build
ubuntu@LINOROBOT:~/YDLidar_SDK.ws/YDLidar-SDK$ ls
build  CMakeLists.txt  csharp  Doxyfile     python     README.pdf  setup.py  startup  ydlidar_config.h.in
cmake  core            doc     LICENSE.txt  README.md  samples     src       test
ubuntu@LINOROBOT:~/YDLidar_SDK.ws/YDLidar-SDK$ cd build
ubuntu@LINOROBOT:~/YDLidar_SDK.ws/YDLidar-SDK/build$ cmake ..
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Current platform: Linux
CMake Deprecation Warning at CMakeLists.txt:50 (cmake_policy):
  The OLD behavior for policy CMP0053 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:53 (cmake_policy):
  The OLD behavior for policy CMP0037 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:56 (cmake_policy):
  The OLD behavior for policy CMP0043 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Found SWIG: /usr/bin/swig4.0 (found version "4.0.2")
-- Found PythonInterp: /usr/bin/python (found version "3.8.13")
-- Found PythonLibs: /usr/lib/aarch64-linux-gnu/libpython3.10.so (found version "3.10.4")
-- Found GTest: /usr/lib/aarch64-linux-gnu/cmake/GTest/GTestConfig.cmake (found version "1.11.0")
CMake Deprecation Warning at core/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at samples/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

[... 68 lines cut ...]
--  Word size (32/64 bit)       : 64
--  CMake version               : 3.22.1
--  CMake generator             : Unix Makefiles
--  CMake build tool            : /usr/bin/gmake
--  Compiler                    : GNU
--  Configuration               :
--
--  __________________________ OPTIONS _____________________________
--  Build YDLidar-SDK as a shared library?　　 :  No
--  Build Examples?　　　　　　　　　　　　　　 : Yes
--  Build C Sharp API?　　　　　　　　　　　　 :  No
--  Build TEST?                         　　　 : Yes
--
--  _________________________ INSTALL _____________________
--  Install prefix                    : /usr/local
--
--  _______________________ WRAPPERS/BINDINGS ______________________
--  Python bindings (pyydlidar)   : Yes
--   - dep: Swig found?       : Yes [Version: 4.0.2]
--   - dep: PythonLibs found?  : Yes [Version: 3.8.13]
--
-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/YDLidar_SDK.ws/YDLidar-SDK/build/CMakeFiles/CMakeOutput.log".

```

JHPHELAN, 2022-09-11:
The directory YDLidar-SDK appears after cloning but has no /build directory
`ubuntu@LINOROBOT:~$ cd YDLidar-SDK/build
bash: cd: YDLidar-SDK/build: No such file or directory`

The directory YDLidar_SDK.ws/YDLidar-SDK does contail /build
However, running cmake.. in this directory gives the following errors:
```
ubuntu@LINOROBOT:~/YDLidar_SDK.ws/YDLidar-SDK/build$ cmake ..
...
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/UseSWIG.cmake:775 (message):
  Policy CMP0078 is not set: UseSWIG generates standard target names.  Run
  "cmake --help-policy CMP0078" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

Call Stack (most recent call first):
  python/CMakeLists.txt:35 (swig_add_library)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/share/cmake-3.22/Modules/UseSWIG.cmake:617 (message):
  Policy CMP0086 is not set: UseSWIG honors SWIG_MODULE_NAME via -module
  flag.  Run "cmake --help-policy CMP0086" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/UseSWIG.cmake:888 (SWIG_ADD_SOURCE_TO_MODULE)
  python/CMakeLists.txt:35 (swig_add_library)
This warning is for project developers.  Use -Wno-dev to suppress it.

Traceback (most recent call last):
  File "<string>", line 2, in <module>
ImportError: cannot import name 'sysconfig' from 'distutils' (/usr/lib/python3.8/distutils/__init__.py)
CMake Error at python/CMakeLists.txt:54 (install):
  install TARGETS given no LIBRARY DESTINATION for module target "_ydlidar".

CMake Error at python/CMakeLists.txt:55 (install):
  install FILES given no DESTINATION!

CMake Warning (dev) at cmake/install_package.cmake:191 (install):
  Policy CMP0062 is not set: Disallow install() of export() result.  Run
  "cmake --help-policy CMP0062" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The file

    /home/ubuntu/YDLidar_SDK.ws/YDLidar-SDK/build/ydlidar_sdkTargets.cmake

  was generated by the export() command.  It should not be installed with the
  install() command.  Use the install(EXPORT) mechanism instead.  See the
  cmake-packages(7) manual for more.

Call Stack (most recent call first):
  CMakeLists.txt:168 (install_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- +===========================================================================+
-- |         Resulting configuration for                             |
-- +===========================================================================+
--  _________________________ PLATFORM _____________________________
--  Host                        : Linux5.15.0-1013-raspiaarch64
--  Is the system big endian?   :  No
--  Word size (32/64 bit)       : 64
--  CMake version               : 3.22.1
--  CMake generator             : Unix Makefiles
--  CMake build tool            : /usr/bin/gmake
--  Compiler                    : GNU
--  Configuration               :
--
--  __________________________ OPTIONS _____________________________
--  Build YDLidar-SDK as a shared library?   :  No
--  Build Examples?               : Yes
--  Build C Sharp API?             :  No
--  Build TEST?                             : Yes
--
--  _________________________ INSTALL _____________________
--  Install prefix                    : /usr/local
--
--  _______________________ WRAPPERS/BINDINGS ______________________
--  Python bindings (pyydlidar)   : Yes
--   - dep: Swig found?       : Yes [Version: 4.0.2]
--   - dep: PythonLibs found?  : Yes [Version: 3.8.13]
--
-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/YDLidar_SDK.ws/YDLidar-SDK/build/CMakeFiles/CMakeOutput.log".
```

JHPHELAN, 2022-10-28:
Now, from U22.04 Jammy / ROS2 Humble when I try the trick of just creating the 'build' directory and running cmake from there, I get a much shorter fatal error message.  Advice??  What's my path to source?  What's my path to build?
```
ubuntu@ubuntu:~/YDLidar_SDK.ws/YDLidar-SDK/build$ cmake
Usage

  cmake [options] <path-to-source>
  cmake [options] <path-to-existing-build>
  cmake [options] -S <path-to-source> -B <path-to-build>

Specify a source directory to (re-)generate a build system for it in the
current working directory.  Specify an existing build directory to
re-generate its build system.

Run 'cmake --help' for more information.
```

The directory prior to adding build is:
```
ubuntu@ubuntu:~/YDLidar_SDK.ws/YDLidar-SDK$ ls -w 1
cmake
CMakeLists.txt
core
csharp
doc
Doxyfile
LICENSE.txt
python
README.md
README.pdf
samples
setup.py
src
startup
test
```

ArchitShahMartian, 2022-11-15:
Thanks a lot! This was really helpful

opteron18, 2022-11-15:
yeah，i creat the /build file and camke ,it really work
 

Opteron
***@***.***

amalshaji4540, 2024-02-24:
> The installation tutorial instructs you to `cd` into `YDLidar-SDK/build` but this directory does not exist in the repository upon cloning. For the sake of new users, add the empty build directory to the repository to circumvent any confusion, since nowhere in the installation instructions does it say to make a new directory.

I also got the same problem