YDLIDAR/YDLidar-SDK #31: make error: (long) unsigned int

JHPHELAN, 2023-02-18:
upon make there are MULTIPLE errors of the form:

>  In member function ‘int ydlidar::core::base::Thread::join(long unsigned int)’:
/home/ubuntu/YDLidar-SDK/./core/base/thread.h:118:18: warning: format ‘%X’ expects argument of type ‘unsigned int’, but argument 2 has type ‘_size_t’ {aka ‘long unsigned int’} [-Wformat=]
  118 |       printf("0x%X thread has been canceled\n", this->_handle);
      |                 ~^                              ~~~~~~~~~~~~~
      |                  |                                    |
      |                  unsigned int                         _size_t {aka long unsigned int}
      |                 %lX

ghdic, 2023-03-05:
I am also getting error in `make`.
this is my configuration

```
-- +===========================================================================+
-- |         Resulting configuration for                             |
-- +===========================================================================+
--  _________________________ PLATFORM _____________________________
--  Host                        : Darwin22.3.0arm64
--  Is the system big endian?   :  No
--  Word size (32/64 bit)       : 64
--  CMake version               : 3.25.2
--  CMake generator             : Unix Makefiles
--  CMake build tool            : /usr/bin/make
--  Compiler                    : Clang
--  Configuration               :
--
--  __________________________ OPTIONS _____________________________
--  Build YDLidar-SDK as a shared library?　　 :  No
--  Build Examples?　　　　　　　　　　　　　　 : Yes
--  Build C Sharp API?　　　　　　　　　　　　 :  No
--  Build TEST?  			　　　 : Yes
--
--  _________________________ INSTALL _____________________
--  Install prefix                    : /usr/local
--
--  _______________________ WRAPPERS/BINDINGS ______________________
--  Python bindings (pyydlidar)   :  No
--   - dep: Swig found?       :  No [Version: ]
--   - dep: PythonLibs found?  : Yes [Version: 3.11]
--
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/life/Downloads/YDLidar-SDK-1.1.1/build
```

and this is my error log when I written `make`

```
~/Downloads/YDLidar-SDK-1.1.1/build ❯ make
[  3%] Building CXX object CMakeFiles/ydlidar_sdk.dir/core/base/timer.cpp.o
[  7%] Building CXX object CMakeFiles/ydlidar_sdk.dir/core/common/ydlidar_def.cpp.o
[ 11%] Building CXX object CMakeFiles/ydlidar_sdk.dir/core/network/ActiveSocket.cpp.o
In file included from /Users/life/Downloads/YDLidar-SDK-1.1.1/core/network/ActiveSocket.cpp:43:
In file included from /Users/life/Downloads/YDLidar-SDK-1.1.1/core/network/ActiveSocket.h:46:
/Users/life/Downloads/YDLidar-SDK-1.1.1/core/network/SimpleSocket.h:120:16: error: use of undeclared identifier 'SHUT_RD'
    Receives = SHUT_RD, ///< Shutdown passive socket.
               ^
/Users/life/Downloads/YDLidar-SDK-1.1.1/core/network/SimpleSocket.h:121:13: error: use of undeclared identifier 'SHUT_WR'
    Sends = SHUT_WR,    ///< Shutdown active socket.
            ^
/Users/life/Downloads/YDLidar-SDK-1.1.1/core/network/SimpleSocket.h:122:12: error: use of undeclared identifier 'SHUT_RDWR'
    Both = SHUT_RDWR    ///< Shutdown both active and passive sockets.
           ^
/Users/life/Downloads/YDLidar-SDK-1.1.1/core/network/SimpleSocket.h:463:3: error: unknown type name 'SOCKET'
  SOCKET GetSocketDescriptor() {
  ^
/Users/life/Downloads/YDLidar-SDK-1.1.1/core/network/SimpleSocket.h:619:24: error: unknown type name 'SOCKET'
  void SetSocketHandle(SOCKET socket) {
                       ^
/Users/life/Downloads/YDLidar-SDK-1.1.1/core/network/SimpleSocket.h:653:3: error: unknown type name 'SOCKET'
  SOCKET               m_socket;            /// socket handle
  ^
/Users/life/Downloads/YDLidar-SDK-1.1.1/core/network/SimpleSocket.h:668:24: error: field has incomplete type 'struct sockaddr_in'
  struct sockaddr_in   m_stServerSockaddr;  /// server address
                       ^
/Users/life/Downloads/YDLidar-SDK-1.1.1/core/network/SimpleSocket.h:668:10: note: forward declaration of 'ydlidar::core::network::sockaddr_in'
  struct sockaddr_in   m_stServerSockaddr;  /// server address
         ^
/Users/life/Downloads/YDLidar-SDK-1.1.1/core/network/SimpleSocket.h:669:24: error: field has incomplete type 'struct sockaddr_in'
[... 31 lines cut ...]
    Close();
    ^~~~~
/Users/life/Downloads/YDLidar-SDK-1.1.1/core/network/ActiveSocket.cpp:62:18: error: variable has incomplete type 'struct in_addr'
  struct in_addr stIpAddress;
                 ^
/Users/life/Downloads/YDLidar-SDK-1.1.1/core/network/ActiveSocket.cpp:62:10: note: forward declaration of 'in_addr'
  struct in_addr stIpAddress;
         ^
/Users/life/Downloads/YDLidar-SDK-1.1.1/core/network/ActiveSocket.cpp:68:35: error: use of undeclared identifier 'AF_INET'
  m_stServerSockaddr.sin_family = AF_INET;
                                  ^
/Users/life/Downloads/YDLidar-SDK-1.1.1/core/network/ActiveSocket.cpp:70:16: error: use of undeclared identifier 'GETHOSTBYNAME'
  if ((m_pHE = GETHOSTBYNAME(pAddr)) == NULL) {
               ^
/Users/life/Downloads/YDLidar-SDK-1.1.1/core/network/ActiveSocket.cpp:75:9: error: use of undeclared identifier 'h_errno'
    if (h_errno == HOST_NOT_FOUND) {
        ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[2]: *** [CMakeFiles/ydlidar_sdk.dir/core/network/ActiveSocket.cpp.o] Error 1
make[1]: *** [CMakeFiles/ydlidar_sdk.dir/all] Error 2
make: *** [all] Error 2
```

Looking at the source code, it seems that I have to fix the error directly. Am I the only one like that?

JHPHELAN, 2023-03-24:
@mjlee111 
@ghdic 
I found that, although there are unsigned int errors in 'make', it didn't fail.
Also 'sudo make install' installs ok.

beautyfree, 2023-06-22:
> sudo make install

same errors on my side

beautyfree, 2023-06-22:
> same errors on my side

> I am also getting error in `make`. this is my configuration
>
> ```
> -- +===========================================================================+
> -- |         Resulting configuration [...]

same errors on my side

JHPHELAN, 2024-01-15:
Having upgraded to Ubuntu 22.04 and ROS2 Humble have re-attempted install and get the same "long unsigned int" warnings, but does not crash.  Would be fantastic if a better coder that I (a very low bar indeed) would update the code to eliminate these warnings before they become errors with the next upgrade.