YDLIDAR/YDLidar-SDK #33: Not able to build and install anything, nothing works in arm64, amd64 or windows system. I need just python library which is also I am not able to build

delhiiitian, 2023-04-07:
Processing /root/YDLidar-SDK
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: ydlidar
  Building wheel for ydlidar (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [213 lines of output]
      running bdist_wheel
      running build
      running build_ext
      build:  build/temp.linux-x86_64-3.10
      CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
        Compatibility with CMake < 2.8.12 will be removed from a future version of
        CMake.

        Update the VERSION argument <min> value or use a ...<max> suffix to tell
        CMake that the project does not need compatibility with older versions.

      -- Current platform: Linux
      CMake Deprecation Warning at CMakeLists.txt:50 (cmake_policy):
        The OLD behavior for policy CMP0053 will be removed from a future version
        of CMake.

        The cmake-policies(7) manual explains that the OLD behaviors of all
        policies are deprecated and that a policy should be set to OLD only under
        specific short-term circumstances.  Projects should be ported to the NEW
        behavior and not rely on setting a policy to OLD.

      CMake Deprecation Warning at CMakeLists.txt:53 (cmake_policy):
        The OLD behavior for policy CMP0037 will be removed from a future version
        of CMake.

        The cmake-policies(7) manual explains that the OLD behaviors of all
        policies are deprecated and that a policy should be set to OLD only under
        specific short-term circumstances.  Projects should be ported to the NEW
        behavior and not rely on setting a policy to OLD.

      CMake Deprecation Warning at CMakeLists.txt:56 (cmake_policy):
        The OLD behavior for policy CMP0043 will be removed from a future version
        of CMake.

        The cmake-policies(7) manual explains that the OLD behaviors of all
        policies are deprecated and that a policy should be set to OLD only under
        specific short-term circumstances.  Projects should be ported to the NEW
        behavior and not rely on setting a policy to OLD.

      CMake Deprecation Warning at core/CMakeLists.txt:1 (cmake_minimum_required):
        Compatibility with CMake < 2.8.12 will be removed from a future version of
        CMake.

        Update the VERSION argument <min> value or use a ...<max> suffix to tell
        CMake that the project does not need compatibility with older versions.

      CMake Error at src/CMakeLists.txt:12 (add_subdirectory):
        The source directory

          /root/YDLidar-SDK/src/core

        does not contain a CMakeLists.txt file.

      CMake Error at src/CMakeLists.txt:12 (add_subdirectory):
        The source directory

          /root/YDLidar-SDK/src/python

        does not contain a CMakeLists.txt file.

      CMake Error at src/CMakeLists.txt:12 (add_subdirectory):
[... 359 lines cut ...]
        File "/usr/lib/python3.10/distutils/dist.py", line 985, in run_command
          cmd_obj.run()
        File "/usr/lib/python3.10/distutils/command/build.py", line 135, in run
          self.run_command(cmd_name)
        File "/usr/lib/python3.10/distutils/cmd.py", line 313, in run_command
          self.distribution.run_command(command)
        File "/usr/lib/python3.10/distutils/dist.py", line 985, in run_command
          cmd_obj.run()
        File "/root/YDLidar-SDK/setup.py", line 36, in run
          self.build_extension(ext)
        File "/root/YDLidar-SDK/setup.py", line 67, in build_extension
          subprocess.check_call(['cmake', ext.sourcedir] + cmake_args, cwd=self.build_temp, env=env)
        File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
          raise CalledProcessError(retcode, cmd)
      subprocess.CalledProcessError: Command '['cmake', '/root/YDLidar-SDK', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/root/YDLidar-SDK/build/lib.linux-x86_64-3.10', '-DPYTHON_EXECUTABLE=/usr/bin/python3', '-DCMAKE_BUILD_TYPE=Release']' returned non-zero exit status 1.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> ydlidar

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.

OtherBob, 2024-04-08:
Same issue, got three units (DFR1030, GS2).  Works on windows but I need to run from RPi4.  Brand new install Debian Bookwork latest packages.  Biggest issue up to this point was that you need to create a VENV for non standard packages.  It would be great if you could provide a wheel through normal sources, but any solutions that gets me running will be acceptable.

sanjana1999, 2024-05-17:
> Same issue, got three units (DFR1030, GS2). Works on windows but I need to run from RPi4. Brand new install Debian Bookwork latest packages. Biggest issue up to this point was that you need to [...]

Facing the same error. Any solutions?