YDLIDAR/YDLidar-SDK #38: Tmini pro: Errors during tri_test and checksum validation

Kamorei, 2023-06-26:
## Environment
- CMake version: 3.22.1
- OS: Ubuntu 22.04
- CPU: AMD Ryzen 5 3500U

## Error Details
I am trying to use Tmini pro. I was able to successfully build and install the latest version of this SDK. However, when I ran tri_test to perform a test, I encountered the following error:

```
__   ______  _     ___ ____    _    ____  
\ \ / /  _ \| |   |_ _|  _ \  / \  |  _ \ 
 \ V /| | | | |    | || | | |/ _ \ | |_) | 
  | | | |_| | |___ | || |_| / ___ \|  _ <  
  |_| |____/|_____|___|____/_/   \_\_| \_\ 

[0] ydlidar /dev/ttyS9
[1] ydlidar2.3 /dev/ttyUSB0
Please select the lidar port:1
Baudrate:
[0] 115200
[1] 128000
[2] 150000
[3] 153600
[4] 230400
[5] 460800
[6] 512000
Please select the lidar baudrate:4
Whether the Lidar is one-way communication [yes/no]:yes
[YDLIDAR] SDK initializing
[YDLIDAR] SDK has been initialized
[YDLIDAR] SDK Version: 1.1.14
[YDLIDAR] Lidar successfully connected /dev/ttyUSB0[230400]
[YDLIDAR] Lidar running correctly! The health status: good
[YDLIDAR] Lidar init success, Elapsed time 653 ms
[YDLIDAR] Start to getting intensity flag
[YDLIDAR] End to getting intensity flag
[YDLIDAR] Create thread 0xA9BFF640
[YDLIDAR] Successed to start scan mode, Elapsed time 2639 ms
[YDLIDAR DEBUG] Thread [0xA9BFF640] ready to cancel[0]
[YDLIDAR DEBUG] Thread [0xA9BFF640] ready to cancel[0] time[0]
[YDLIDAR] Thread [0xA9BFF640] has been canceled
[YDLIDAR] Failed to turn on the Lidar, because the lidar is [No error].
Fail to start Unknown error
```

I also sometimes experience a checksum error like this:

```
__   ______  _     ___ ____    _    ____  
\ \ / /  _ \| |   |_ _|  _ \  / \  |  _ \ 
 \ V /| | | | |    | || | | |/ _ \ | |_) | 
  | | | |_| | |___ | || |_| / ___ \|  _ <  
  |_| |____/|_____|___|____/_/   \_\_| \_\ 

[0] ydlidar /dev/ttyS9
[1] ydlidar2.3 /dev/ttyUSB0
Please select the lidar port:1
Baudrate:
[0] 115200
[1] 128000
[2] 150000
[3] 153600
[4] 230400
[5] 460800
[6] 512000
Please select the lidar baudrate:4
Whether the Lidar is one-way communication [yes/no]:yes
[YDLIDAR] SDK initializing
[YDLIDAR] SDK has been initialized
[YDLIDAR] SDK Version: 1.1.14
[YDLIDAR] Lidar successfully connected /dev/ttyUSB0[230400]
[YDLIDAR] Lidar running correctly! The health status: good
[YDLIDAR] Lidar init success, Elapsed time 651 ms
[YDLIDAR] Start to getting intensity flag
[YDLIDAR] End to getting intensity flag
[YDLIDAR] Create thread 0xD29FF640
[YDLIDAR] Checksum error c[0x6C] != r[0x06]
[YDLIDAR] Successed to start scan mode, Elapsed time 2299 ms
[YDLIDAR] Checksum error c[0x94] != r[0x36]
[YDLIDAR DEBUG] Thread [0xD29FF640] ready to cancel[0]
[YDLIDAR DEBUG] Thread [0xD29FF640] ready to cancel[0] time[0]
[YDLIDAR] Thread [0xD29FF640] has been canceled
[YDLIDAR] Failed to turn on the Lidar, because the lidar is [No error].
Fail to start Unknown error
```

## Additional Info
Based on past experience, I know that it works fine with the commit made on Feb 21, 2023 (2317c4def8ac26d49d8b79b76d2be89f2c3bcf79).

esamet, 2023-07-10:
yes ı had a same error, Did you resolve that?

Kamorei, 2023-07-11:
No, I have already applied the latest commit from July 4 (1551f47430679be3f53dca82b64c7588ed927024), but I am still encountering the same error.
Currently, I am using an older commit (2317c4def8ac26d49d8b79b76d2be89f2c3bcf79) and it works without any issues.

yos-mur, 2023-08-20:
I used latest version 1.1.1 on linux (debian bullseye) and same error was occur. 
But "tmini_test" seems to work fine.

Kamorei, 2023-08-27:
When built with version 1.1.1, tmini_test was not found.
However, in previous commit  (2317c4def8ac26d49d8b79b76d2be89f2c3bcf79) and the latest commit (1140d7468dcfb7cca9f281ff227d88ed19781c79),  tmini_test exists and it seems to work fine.
This test program might be more accurate for the Tmini Pro.

akela1101, 2023-10-12:
Version `1.1.19` works for me on Ubuntu 22 (armv8) through ros2 driver. But `tri_test` fails as soon as it gets checksum error, so I guess it's just a bug in tri_test.