YDLIDAR/YDLidar-SDK #63: Ubuntu 24.04 on Pi5 - Tmini_test got  Check Sum xxxxx

mrRobot62, 2025-03-30:
- Environment: Ubuntu 24.04 on PI5
- lastest YDlidar-SDK Version
- No compile errors
- YDLidar TMiniPlus

Szenario:
from 'build' folder starting ./tmini_test

ends with
```
bernd@ros2pi5:~/ros2_ws/src/YDLidar-SDK/build$ ./tmini_test
[2025-03-30 15:31:48][info] __   ______  _     ___ ____    _    ____
[2025-03-30 15:31:48][info] \ \ / /  _ \| |   |_ _|  _ \  / \  |  _ \
[2025-03-30 15:31:48][info]  \ V /| | | | |    | || | | |/ _ \ | |_) |
[2025-03-30 15:31:48][info]   | | | |_| | |___ | || |_| / ___ \|  _ <
[2025-03-30 15:31:48][info]   |_| |____/|_____|___|____/_/   \_\_| \_\
[2025-03-30 15:31:48][info] 
[2025-03-30 15:31:48][info] SDK initializing
[2025-03-30 15:31:48][info] SDK has been initialized
[2025-03-30 15:31:48][info] SDK Version: 1.2.12
[2025-03-30 15:31:48][info] Stop Lidar
[2025-03-30 15:31:48][info] Connect elapsed time 536 ms
[2025-03-30 15:31:48][info] Lidar successfully connected [/dev/ttyUSB0:230400]
[2025-03-30 15:31:48][info] Lidar running correctly! The health status good
[2025-03-30 15:31:49][info] Current Lidar Model Code 151
[2025-03-30 15:31:49][info] Baseplate device info
Firmware version: 1.2
Hardware version: 1
Model: Tmini Plus
Serial: 2024092400090093
[2025-03-30 15:31:49][info] Current scan frequency: 10.00Hz
[2025-03-30 15:31:49][info] Current scan frequency: 10.00Hz
[2025-03-30 15:31:49][info] Check status, Elapsed time 356 ms
[2025-03-30 15:31:49][info] Lidar init success, Elapsed time [892]ms
[2025-03-30 15:31:49][warn] Fail to get pitch angle
[2025-03-30 15:31:51][info] [YDLIDAR] Create thread 0x8EA0F160
[2025-03-30 15:31:51][error] Check Sum 0x8D5C != 0x6758
[2025-03-30 15:31:51][error] Check Sum 0x17F2 != 0x5DB9
[2025-03-30 15:31:51][error] Check Sum 0x0BD8 != 0x7178
[2025-03-30 15:31:51][error] Check Sum 0xA751 != 0x52FB
[2025-03-30 15:31:51][error] Check Sum 0x3512 != 0x97F8
[2025-03-30 15:31:51][error] Check Sum 0x4227 != 0xE893
[2025-03-30 15:31:51][error] Check Sum 0xAFCE != 0x413B
[2025-03-30 15:31:51][error] Check Sum 0x8627 != 0x114A
[2025-03-30 15:31:51][error] Check Sum 0xE843 != 0x67D3
[2025-03-30 15:31:51][error] Check Sum 0x3A14 != 0x504B
```

Scanning stopped after ~10secs

BTW:
I wrote a PythonScript and implemented the checksum algorithm descriped in documentation, but I get checksum error too.

Any help? Any assistance ? Any ideas?

Regards Bernd

mrRobot62, 2025-03-30:
Testing with tri_test

```
bernd@ros2pi5:~/ros2_ws/src/YDLidar-SDK/build$ ./tri_test
__   ______  _     ___ ____    _    ____  
\ \ / /  _ \| |   |_ _|  _ \  / \  |  _ \ 
 \ V /| | | | |    | || | | |/ _ \ | |_) | 
  | | | |_| | |___ | || |_| / ___ \|  _ <  
  |_| |____/|_____|___|____/_/   \_\_| \_\ 

Baudrate:
[0] 115200
[1] 128000
[2] 150000
[3] 153600
[4] 230400
[5] 460800
[6] 512000
Please select the lidar baudrate:4
Whether the Lidar is one-way communication [yes/no]:yes
[2025-03-30 15:35:03][info] SDK initializing
[2025-03-30 15:35:03][info] SDK has been initialized
[2025-03-30 15:35:03][info] SDK Version: 1.2.12
[2025-03-30 15:35:03][info] Connect elapsed time 24 ms
[2025-03-30 15:35:03][info] Lidar successfully connected [/dev/ttyUSB0:230400]
[2025-03-30 15:35:03][info] Lidar running correctly! The health status good
[2025-03-30 15:35:03][error] Fail to get baseplate device information!
[2025-03-30 15:35:03][info] Check status, Elapsed time 0 ms
[2025-03-30 15:35:03][info] Lidar init success, Elapsed time [24]ms
[2025-03-30 15:35:04][info] Start to getting intensity flag
[2025-03-30 15:35:04][info] Auto set intensity 1
[2025-03-30 15:35:04][info] [YDLIDAR] End to getting intensity flag
[2025-03-30 15:35:04][info] [YDLIDAR] Create thread 0x9EA0F160
[2025-03-30 15:35:05][info] Successed to start scan mode, Elapsed time 1723 ms
[2025-03-30 15:35:05][info] Scan Frequency: 5.00Hz
[2025-03-30 15:35:05][info] Fixed Size: 720
[2025-03-30 15:35:05][info] Sample Rate: 4.00K
[2025-03-30 15:35:05][info] Scan Frequency: 5.00Hz
[2025-03-30 15:35:05][info] Fixed Size: 720
[2025-03-30 15:35:05][info] Sample Rate: 4.00K
[2025-03-30 15:35:05][info] Scan Frequency: 5.00Hz
[2025-03-30 15:35:05][info] Fixed Size: 720
[2025-03-30 15:35:05][info] Sample Rate: 4.00K
[2025-03-30 15:35:05][info] Single Fixed Size: 320
[2025-03-30 15:35:05][info] Sample Rate: 4.00K
[2025-03-30 15:35:05][info] Successed to check the lidar, Elapsed time 251 ms
[2025-03-30 15:35:05][info] Now lidar is scanning...
Fail to get module device info
Fail to get baseplate device info
Scan received [332] points scanFreq [12.60]
Time consuming [2080] from initialization to parsing to point cloud data
Scan received [340] points scanFreq [12.30]
Scan received [352] points scanFreq [11.90]
...
```

no errors, test run until CTRL-C

mrRobot62, 2025-03-30:
tof_test ends with Checksum-Error:

```
bernd@ros2pi5:~/ros2_ws/src/YDLidar-SDK/build$ ./tof_test
__   ______  _     ___ ____    _    ____  
\ \ / /  _ \| |   |_ _|  _ \  / \  |  _ \ 
 \ V /| | | | |    | || | | |/ _ \ | |_) | 
  | | | |_| | |___ | || |_| / ___ \|  _ <  
  |_| |____/|_____|___|____/_/   \_\_| \_\ 

Baudrate:
0. 115200
1. 230400
2. 460800
3. 512000
Please select the lidar baudrate:1
Whether the Lidar is one-way communication[yes/no]:yes
[2025-03-30 15:37:22][info] SDK initializing
[2025-03-30 15:37:22][info] SDK has been initialized
[2025-03-30 15:37:22][info] SDK Version: 1.2.12
[2025-03-30 15:37:22][info] Connect elapsed time 24 ms
[2025-03-30 15:37:22][info] Lidar successfully connected [/dev/ttyUSB0:230400]
[2025-03-30 15:37:22][info] Lidar running correctly! The health status good
[2025-03-30 15:37:22][error] Fail to get baseplate device information!
[2025-03-30 15:37:22][info] Check status, Elapsed time 0 ms
[2025-03-30 15:37:22][info] Lidar init success, Elapsed time [24]ms
[2025-03-30 15:37:22][info] [YDLIDAR] Create thread 0xADC0F160
[2025-03-30 15:37:22][error] Check Sum 0x11B9 != 0x2292
[2025-03-30 15:37:22][error] Check Sum 0x9F22 != 0x07E7
[2025-03-30 15:37:22][error] Check Sum 0x2C14 != 0x7DF2
...
```

mira-lily, 2025-04-04:
I have also same error with ROS2_driver. Anyone help to fix this error, please?